crycode-de/mcp-can-boot #1: Not working on Arduino Mini ?

md-tec, 2021-05-23:
I flash the boatloader via PIO and an Arduino Mega as ISP. But on my RPI i dont see any Action on CAN Bus. The flash App will also not work. Can you give an Example how to flash an Arduino Nano 328p over CAN and How to install the Bootloader on it ?

Thanks.

crycode-de, 2021-05-24:
The Arduino Nano 328p should work fine, as the ATmega328P is fully supported by MCP-CAN-Boot.

You have to connect the MCP2515 with the Arduino as follows:
| MCP2515 | Arduino |
|---|---|
| VCC | 5V |
| GND | GND |
| CS | D10 |
| MISO | D12 |
| MOSI | D11 |
| SCK | D13 |

For some visual output you may attach a LED to your Arduino, e.g. on D7 and add this to `config.h`:
```c
#define LED      PORTD7
#define LED_DDR  DDRD
#define LED_PORT PORTD
```

With the LED enabled, if there is an error while initializing the MCP2515 the LED should flash in 50ms impulses.

md-tec, 2021-05-27:
I connected a LED to D7 and edited the Configfile. After Upload via PIO with an Arduino Mega as ISP the LED is always off. Is it possible that the Bootloader is not flashed sucessuflly ?

crycode-de, 2021-05-27:
The LED should light up for the configured [timeout](https://github.com/crycode-de/mcp-can-boot/blob/e125c1a211e1ed193002cfd135bffbc24ee80ff9/src/config.h#L39) (if no flashing via CAN is started in this time).
If the LED is always off, there must be an other issue.
If flashing the bootloader was not sucessfull, PIO should show an error since it verifys the flashed data.

Did you set the correct fuse bits?
The default High Fuse of an Arduino Nano is `0xDA` as far as I now, which indicates a 1024 words bootloader section.
MCP-CAN-Boot requires a 2048 words bootloader section, so the High Fuse has to be `0xD8`.

md-tec, 2021-05-28:
I used the env:ATmega328P from the platformio.ini which comes with this Project.. I uncommented the follwing Lines:

```
board_fuses.lfuse = 0xFF
board_fuses.hfuse = 0xD8
board_fuses.efuse = 0xFC
```

The  env:ATmega328P is:

```
[platformio]
;default_envs = ATmega1284P

[env]
platform = atmelavr
framework = arduino

build_flags =
  -Os ; optimize for size
  -funsigned-char
  -funsigned-bitfields
  -fpack-struct
  -fshort-enums
  -fno-exceptions
  -fno-jump-tables
  -mrelax
  -Wl,--relax
  -Wl,--cref
  -Wl,-Map=$BUILD_DIR/firmware.map

board_build.f_cpu = 16000000L

upload_protocol = stk500v1
upload_flags =
  -P$UPLOAD_PORT
  -b$UPLOAD_SPEED
  -e

upload_port = COM10
upload_speed = 19200

[env:ATmega328P]
board = ATmega328P
build_flags =
  ${env.build_flags}
  -Wl,--section-start=.text=0x7000 ; 2048 words bootloader, 0x3800 * 2

board_fuses.lfuse = 0xFF
board_fuses.hfuse = 0xD8
board_fuses.efuse = 0xFC
```

PIO says successful when the Upload is Done. Is there an alternative Way to flash the Bootloader ?

crycode-de, 2021-05-28:
The fuses are not touched by PIO by default.
To set them according to the config in `platformio.ini` try to run `pio run --target fuses` from a terminal.

You may also use the hex file created by PIO and flash it with any other application you like, but the result will be the same imho.

Alternatively you may set the fuses manually using avrdude:
`avrdude -p m328p -c arduino -P COM10 -b 19200 -v -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xFC:m`

md-tec, 2021-05-30:
It is Working 👍  Thanks for your Help. I didnt know that i had to run` pio run --target fuses` from Terminal first. 

Thanks for the quick help.

closed 2021-05-30 by md-tec

crycode-de, 2021-05-31:
Thank you for your reply.

I've added some information for this in the [readme](https://github.com/crycode-de/mcp-can-boot#flashing-the-bootloader).