crycode-de/mcp-can-boot #4: Bootloader on ATMEGA64M1

alessio31183, 2021-12-01:
I'm wondering if, using a different driver file instead of mcp2515.cpp and mcp2515.h, I can port the project on the ATMEGA64M1.

That's my uC used on all my canBus project.

How can I proceed?

Because I don't see the .hex file for programming

Thanks in advance

crycode-de, 2021-12-01:
In general it should be possible to replace all calls to the `mcp2515` instance in `bootloader.cpp` by the commands needed to interact with the buildin CAN controller of the ATmage64M1 and strip out all `mcp2515` calls.

Also you have to add the µC in `controllers.h` with the correct definitions.

Then you should be able to build the .hex file for your µC.

alessio31183, 2021-12-02:
Ok Peter Thanks,

I'll try to substitute drivers for the ATMEGA64M1 in order to fit the bootloader with my uC.

One question: for the remote flash application which kind of CAN interface I can use?

Can I use an Arduino with the can controller shield with the MCP2515 on board?

In case yes, how can I tell it to the flash application?

Thanks

crycode-de, 2021-12-02:
Currently the flash application needs a Linux system with a CAN interface supported by the kernel.
A lightweight solution is a Raspberry Pi with a MCP2515.

I had some plans to support an Arduino with a MCP2515, but no time to implement it yet.

alessio31183, 2021-12-03:
I use Ubuntu with MCP-CAN-Boot Flash-App installed.

I'm trying to add the kernel support to this usb-to-can converter https://github.com/alexmohr/usb-can.

crycode-de, 2021-12-03:
Looks good. If you get the `can0` interface working, the flash application should do the job 👍

closed 2022-07-05 by crycode-de