crycode-de/mcp-can-boot #6: Works like charm! Thanks a lot!

Nerdiyde, 2022-07-01:
Hi Peter,
thanks a lot for sharing this. Its exactly what I was looking for.

I'm pretty new to platform.io and CAN on AVR and Raspberry Pi. So it took me some hours to make all of this run properly.

I just thought I could contribute some of my experiences I had during the setup process. Maybe this makes it easier for others to use this.

In my environment I'm using an atmega328p and an AVR ISP MK2 for flashing. 
The following platformio settings worked for me

```
[env:ATmega328P_via_AVRISP_mkII]
board = ATmega328P
build_flags =
  ${env.build_flags}
  -Wl,--section-start=.text=0x7000 ; 2048 words bootloader, 0x3800 * 2

board_build.f_cpu = 16000000L

board_fuses.lfuse = 0xFF
board_fuses.hfuse = 0xD8
board_fuses.efuse = 0xFC

platform = atmelavr
framework = arduino
upload_protocol = custom
upload_port = usb
upload_flags =
    -C
    ; use "tool-avrdude-megaavr" for the atmelmegaavr platform
    ${platformio.packages_dir}/tool-avrdude/avrdude.conf
    -p
    $BOARD_MCU
    -P
    $UPLOAD_PORT
    -c
    stk500v2 ; avrispmkII
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i

; to flash the fuses run "platformio.exe run --target fuses --environment ATmega328P_via_AVRISP_mkII"
```

Maybe you can/want to add it to the platformio.ini file?

Additionally I used successfully this CAN adapter on the raspberr pi: 

https://www.amazon.de/Konverter-Raspberry-Computer-Support-USB2CAN-C/dp/B09K3LL93Q/
Together with the helpfull infos from your page here: https://crycode.de/can-bus-am-raspberry-pi

Additionally I had to update node js on my pi to version 12.
For this I used NVM: https://github.com/nvm-sh/nvm
Executed this script on my pi: curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.39.1/install.sh | bash
To install nodejs 12 using this command: nvm install 12 (make sure to perform a relogin into your bash otherwise the command nvm will not be known after install of nvm)

After that I was able to follow the instructions here: https://github.com/crycode-de/mcp-can-boot-flash-app
to install the flash app.

Some other thing I noticed:
First I tried to flash a huge hex file (about 29214 bytes). Of course this didnt fit into the flash together with the bootloader. The app on the raspi showed as error message 
"Flash address error!                                                                                                                                                                                                                         Maybe the hex file is not for this MCU type?"
Maybe its possible to add here an additional hint that this could also be caused by a to large application? :)

Nevertheless, awesome tool and thanks again for sharing!

Ah btw. do you mind if I try to port the flash application to an ESP32? In my usecase I have an ESP32 on the same CAN bus together with some other AVRs and (just for in case) it would be awesome to flash the AVRs via the ESP32.

Thanks and best regards
Fabian

crycode-de, 2022-07-05:
Thank you for your feedback!
I've added the env to the `platformio.ini` file and updated the "Flash address error" message in the flash application.

Feel free to fork and port the bootloader and/or the flash application to other platforms according to the _CC BY-NC-SA 4.0_ license.
Maybe we could merge this back to this project and name it simply "CAN-Boot". 🙂

crycode-de, 2022-07-14:
Great idea!

I took a quick look at it. Looks good so far. :)

Think about implementing the "ping" messages, which is usefull for the bitrate-detect feature (branch [bitrate-detection](https://github.com/crycode-de/mcp-can-boot-flash-app/tree/bitrate-detection), see [compare](https://github.com/crycode-de/mcp-can-boot-flash-app/compare/bitrate-detection)).

Please let me know, when you think you're done. Then I'll add a link to you repo. ;)

Nerdiyde, 2022-07-15:
Hi,

Thanks. :)
just added the ping message impelementation. Please see it here: https://github.com/Nerdiyde/AVR_CAN_flasher/commit/f53be1548046348bc8c8d85ecae23d283ea2645f

Sadly I didnt had the time to test all the features yet. Especially when it comes ot read hex files and save them locally. 
But the basic functionality (flashing) is fine from my point of view. Alreading integrated this in my setup and used it the last days. :)

Best regards
Fab

commit bffb3c32 (2023-06-04): Updated readme and some code comments

Closes #6
closed 2023-06-04 by crycode-de