crycode-de/mcp-can-boot #9: Flash data error! Maybe there are some CAN bus issues?

Chemikhan, 2023-06-19:
Hello Peter!

Firstly, I would like to thank you for this project. I think that it is very helpful for many people.

However, I would like to ask you for help at solving one problem with flashing Arduino Nano (ATmega328P) with *.hex file.
Maybe I'll go into too much detail, but I think it will help other guys who are facing the same problems to solve them faster.
Arduino Nano(MCU) and RPi (Flasher) are connected through the MCP2515

1) I installed Arduino as ISP sketch on Arduino Uno and connected it to the Arduino Nano:

| UNO| -| NANO|
|--------|--------|--------|
| D13 | --> | D13 |
| D12 | --> | D12 |
| D11 | --> | D11 |
| D10 | --> | RST |
|  5V  | --> | 5V  | 
| GND| --> | GND| 

2) I connected one led to the D7 and added to the config.h:
```
#define LED      PORTD7
#define LED_DDR  DDRD
#define LED_PORT PORTD
```

3) Then I uploaded bootloader.cpp (as I understand) to the Arduino Nano by the "Upload" button in the Project Tasks of the PlatformIO project. 
[image]
In fact, the command looks like: 
`C:\Users\USER\.platformio\penv\Scripts\platformio.exe run --target upload --environment ATmega328P 
`
4) The LED is not blinking.  

5) I set fuses using button from Project Tasks:
[image]
In fact, the command looks like: 
`C:\Users\USER\.platformio\penv\Scripts\platformio.exe run --target fuses --environment ATmega328P
`...
Selected fuses: [lfuse = 0xFF, hfuse = 0xD8, efuse = 0xFC] 
...

6) Nothing happens with LED.  
8) Output of the command on the Flasher side
`npx mcp-can-boot-flash-app -f firmware.hex -p m328p -m 0x0042`
is 
`Waiting for bootloader start message for MCU ID 0x0042 ...`
And nothing happens.

7)  I upload bootloader again (p.3). The led is blinking 3 times in a row with small pause, then comes bigger pause and whole process repeats. Looks like it's waiting for flashing.

8) Output of the command on the Flasher side
`npx mcp-can-boot-flash-app -f firmware.hex -p m328p -m 0x0042`
is different every time but everytime flashing stops around 10-15% after several messages 
"Flash data error! 
Maybe there are some CAN bus issues?"
[image]

9) I can not start the command again till I restart CAN bus. Also, "ifconfig" command on the Flasher side shows that there is a TX problem on the CAN bus. After restarting CAN bus, I can run the command again but the result is the same as in p.8.

Could you please tell me, what I am doing wrong?

crycode-de, 2023-06-20:
Hi,
your wiring and LED config looks ok.

Did you set the correct `MCP_CLOCK` value according to your hardware?

Did you enabled the bitrate detect feature, or are you using a fixed bitrate?

lfuse = 0xFF, hfuse = 0xD8, efuse = 0xFC should be fine for the Arduino Nano.

Generally you should flash the bootloader after setting the correct fuse bits.

The "Setting flash address to ..." Message indicates that the flash app and the bootloader are out of sync for some reason. Normally this message should not appear since the bootloader always responds with the next flash address which matches the expected address from the flash app.

As you see, some messages are send successfully over the bus, but some are not.

Probably you have a hardware issue in your CAN hardware.
What are you using exactly?
Is the bus (CAN H and CAN L) terminated at both ends?

Chemikhan, 2023-06-21:
The problem was in termination resistors. Now everything works. Thank you very much!

closed 2023-06-21 by Chemikhan