crycode-de/mcp-can-boot #13: Different Value Of Extended Fuse Required On Atmel

ptoshkov, 2024-02-10:
Dear all,

Thank you for the lovely project you have created.

My device is Arduino Uno R3 with microcontroller Atmel ATmega328P.

With the fuse settings in **_platformio.ini_**, flashing over CAN does not work.

`[env:ATmega328P]
board = ATmega328P
build_flags =
  ${env.build_flags}
  -Wl,--section-start=.text=0x7000 ; 2048 words bootloader, 0x3800 * 2
;board_fuses.lfuse = 0xFF
;board_fuses.hfuse = 0xD8
;board_fuses.efuse = 0xFC`

For the fuse "efuse", I had to use the value "0xFD" for the flashing over CAN to work.

crycode-de, 2024-02-10:
Hi,
the fuse values in `platformio.ini` are only examples and need be adjusted according your hardware.
Therefore they are commented out by default.

The only relevant fuse bits for the bootloader itself are BOOTRST, BOOTSZ0 and BOOTSZ1 in the hfuse byte.

For ATmega328P in the efuse byte only the brown-out detection level is defined.
See [AVR Fuse Calculator](http://eleccelerator.com/fusecalc/fusecalc.php?chip=atmega328p&LOW=FF&HIGH=D8&EXTENDED=FC&LOCKBIT=FF) for reference.

ptoshkov, 2024-02-10:
Ok, makes sense 👍

closed 2024-02-10 by crycode-de