crycode-de/mcp-can-boot #14: Support for ATmega328PB (Arduino Pro Mini)

Baterka, 2024-08-10:
Hello, amazing work on this one.

I have just one small request. Can you please add support for ATmega328PB (Arduino Pro Mini)?

I tried to do it by myself, but even though I am able to upload the bootloader, it does not work. It is working fine on same setup with standard ATmega328P (Arduino Nano), so it is not cause dby bad setup.

**What I tried:**

- Adding this to `platformio.ini`
```
[env:ATmega328PB]
board = ATmega328PB
build_flags =
  ${env.build_flags}
  -Wl,--section-start=.text=0x7000 ; 2048 words bootloader, 0x3800 * 2

; Fuses specific to 328PB I found on internet
board_fuses.lfuse = 0xFF
board_fuses.hfuse = 0xDE
board_fuses.efuse = 0xFD
```

- Editing line 51 in `controllers.h` to this:
```
#elif defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328PB__)
```

But it does not work. It seems that the 328PB have something different from the 328P.

**Why I need this?**
I bought 10 pieces of Pro Mini boards for my Home Automation project but they are 328PB based :/

Thanks for any help!

crycode-de, 2024-08-10:
Hi, I've checked the datasheet of the ATmega328PB and also the application note AT15007 "Differences between ATmega328/P and
ATmega328PB".
From the AN:
> ATmega328PB is not a drop-in replacement for ATmega328 variants, but a
> new device. However, the functions are backward compatible with the
> existing ATmega328 functions. Existing code for these devices will work in
> the new devices without changing existing configuration or enabling new
> functions. The code that is available for your existing ATmega328 variants
> will continue to work on the new ATmega328PB device.

The ports and registers used by the bootloader are the same on both devices.
The relevant fuse bits are also the same.

So your changes should be fine.

But there **is** an issue regarding your hfuse config.
When hfuse is set to 0xDE, then BOOTRST is enabled but BOOTSZ0 and BOOTSZ1 are not. You need to enable them too, which results in hfuse 0xD8 (like for ATmega328P).

If it's working for you, I'll add the ATmega328PB to the code in the repo and to the list of the supported devices. :)

Baterka, 2024-08-11:
> Hi, I've checked the datasheet of the ATmega328PB and also the application note AT15007 "Differences between ATmega328/P and
> ATmega328PB".
> From the AN:
>
> The ports and registers used by the [...]

Thanks for response ;) I retested it today with 328P fuses configuration and it works fine. I am idiot that I did not tried it with those fuses first :D 

When I was in it, I also made changes to the Node.JS uploader code and added support for Windows. I will create PRs on both sides soon ;)

crycode-de, 2024-08-11:
That's perfekt!
Looking forward for your PRs. :)

closed 2024-08-14 by Baterka